watterott/ATmega328PB-Testing #7: avrdude Linux support

woo2, 2016-07-06:
The avrdude binary for linux is missing from this repo. Installation fails because arduino needs a local copy. I am using a current version of Arch Linux with avrdude 6.3 installed system wide. 

Am I doing something wrong? [I added the avrdude binary here.](https://github.com/woo2/ATmega328PB-Testing) Dependencies were not statically linked against it, so it won't work on everyone's system. I will rebuild it and submit a pull request when ready.

awatterott, 2016-07-06:
Currently everything is only for Windows, but Arduino.cc is working on a new toolchain including avrdude 6.3. Further infos here: https://github.com/arduino/Arduino/pull/5021

awatterott, 2016-07-26:
Arduino IDE 1.6.10 includes gcc 4.9.2 and avrdude 6.3.

closed 2016-07-26 by awatterott