watterott/ATmega328PB-Testing #9: Cannot upload Blink sketch to 328PB

paulusjacobus, 2016-09-18:
I have trouble to upload my first sketch into my UNO 328PB (I replaced the 328p smd chip with a 328pb). I managed to burn the optiboot loader via an uno as ISP programmer. The IDE can see my new uno (com port and get board info: 
BN: Unknown board
VID: 1A86
PID: 7523
SN: Upload any sketch to obtain it)
I am stuck within the upload and looking for suggestions what I can do next to get it going.

I get the follow verbose output after hitting "upload" in the IDE:

Arduino: 1.6.12 Hourly Build 2016/09/14 08:20 (Windows 10), Board: "Atmel ATmega328PB Crystal Clock, 16 MHz"

C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware -hardware C:\Users\Paul_2\AppData\Local\Arduino15\packages -tools C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\tools-builder -tools C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -tools C:\Users\Paul_2\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\libraries -libraries C:\Users\Paul_2\Documents\Arduino\libraries -fqbn=arduino:avr:atmega328pbcc:speed=16mhz -ide-version=10612 -build-path C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\examples\01.Basics\Blink\Blink.ino
C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware -hardware C:\Users\Paul_2\AppData\Local\Arduino15\packages -tools C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\tools-builder -tools C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\tools\avr -tools C:\Users\Paul_2\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\libraries -libraries C:\Users\Paul_2\Documents\Arduino\libraries -fqbn=arduino:avr:atmega328pbcc:speed=16mhz -ide-version=10612 -build-path C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\examples\01.Basics\Blink\Blink.ino
Using board 'atmega328pbcc' from platform in folder: C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr
Detecting libraries used...
"C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\cores\arduino" "-IC:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\variants\atmega328pb" "C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\sketch\Blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\cores\arduino" "-IC:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\variants\atmega328pb" "C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\sketch\Blink.ino.cpp" -o "C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\cores\arduino" "-IC:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\variants\atmega328pb" "C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\sketch\Blink.ino.cpp" -o "C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\sketch\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\tools\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10612 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\cores\arduino" "-IC:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\variants\atmega328pb" "C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\hardware\arduino\avr\cores\arduino\wiring_pulse.S" -o "C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\wiring_pulse.S.o"
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\WInterrupts.c.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\hooks.c.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\wiring.c.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\wiring_analog.c.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\wiring_digital.c.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\wiring_pulse.c.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\wiring_shift.c.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\CDC.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\HardwareSerial.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\HardwareSerial0.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\HardwareSerial1.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\HardwareSerial2.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\HardwareSerial3.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\IPAddress.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\PluggableUSB.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\Print.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\Stream.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\Tone.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\USBCore.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\WMath.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\WString.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\abi.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\main.cpp.o
Using previously compiled file: C:\Users\Paul_2\AppData\Local\Temp\arduino_build_916475\core\new.cpp.o
[... cut ...]
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 8 of 10: not in sync: resp=0x11
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 9 of 10: not in sync: resp=0x11
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0x11

avrdude.exe done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

awatterott, 2016-09-18:
Looks like to bootloader is not correctly flashed. Can you send the log messages from the bootloader burning and have you used the [hex file](https://github.com/watterott/ATmega328PB-Testing/raw/master/hardware/atmega328pb/avr/bootloaders/optiboot_m328pb.hex) from this repo?

paulusjacobus, 2016-09-18:
Yes I used all the files and intel hex bootloader file from this repo. Tomorrow I will reflash the 328pb again and send you the log messages.

paulusjacobus, 2016-09-18:
These are the log messages:

C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino2/bin/avrdude.exe -CC:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5/tools/avrdude.conf -v -patmega328pb -cstk500v1 -PCOM7 -b19200 -e -Ulock:w:0xFF:m -Uefuse:w:0xF5:m -Uhfuse:w:0xDC:m -Ulfuse:w:0xFF:m 

avrdude.exe: Version 6.3, compiled on Jun 22 2016 at 16:05:21
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5/tools/avrdude.conf"

             Using Port                    : COM7
             Using Programmer              : stk500v1
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega328PB
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude.exe: Device signature = 0x1e9516 (probably m328pb)
avrdude.exe: erasing chip
avrdude.exe: reading input file "0xFF"
avrdude.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 1 bytes of lock written
avrdude.exe: verifying lock memory against 0xFF:
avrdude.exe: load data lock data from input file 0xFF:
avrdude.exe: input file 0xFF contains 1 bytes
avrdude.exe: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s
[... 116 lines cut ...]
avrdude.exe: load data flash data from input file C:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5/bootloaders/optiboot_m328pb.hex:
avrdude.exe: input file C:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5/bootloaders/optiboot_m328pb.hex contains 32748 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verifying ...
avrdude.exe: 32748 bytes of flash verified
avrdude.exe: reading input file "0xCF"
avrdude.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude.exe: 1 bytes of lock written
avrdude.exe: verifying lock memory against 0xCF:
avrdude.exe: load data lock data from input file 0xCF:
avrdude.exe: input file 0xCF contains 1 bytes
avrdude.exe: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude.exe: verifying ...
avrdude.exe: 1 bytes of lock verified

avrdude.exe done.  Thank you.

awatterott, 2016-09-18:
This looks okay. I also tried the latest hourly IDE build and it works also. Have you tried the IDE v1.6.11?
There was a small problem with the bootloader area size, which was set to 512 words (1024 bytes), but this cannot lead to your issues. It is fix in v1.0.6.

paulusjacobus, 2016-09-19:
Hi Andreas, I have tried IDE v1.6.11 and 12. The only thing that is odd
when programming and uploading is the different baudrate that the IDE
reports back to me. The bootloader is burned at 19200 baud but the uploader
tries a higher baud rate i.e 58000. Could that be the issue? How do I
change the uploader baudrate? I thought you set this in AVRdude.conf but it
seems to be ignored.
Also when trying to upload it gets a one byte response from the UNO board
which seems to be random 0x11 or 0xCF etc.
The com port is available in the IDE but when trying to get the board info:
the board is unknown. Shouldn't is say atmega328pbcc or something like
that? If the board cannot be reported back, would that stop the upload?
Sorry for the many questions, but I am fairly new to compilers and cross
linkers. Thanks for your help, it is much appreciated.

paulusjacobus, 2016-09-19:
The boards.txt file was correct and setting the baudrate did not make any difference. Wonder if i had the correct place? what are the fuse settings? I use a standard uno r3 board so the crystal osc driven mcu choice should be the one i need i guess. So i am a bit stuck now.

awatterott, 2016-09-19:
The different baudrates for the ISP programmer and bootloader are correct.
About the board info: the detection only sees the USB serial adapter (ATmega16u2) and its USB id. It does not recognize the ATmega328 microcontroller.
Our bootloader has an automatic baudrate detection and so the crystal speed and baudrate is not relevant. However you cannot see if the bootloader is active. The standard/original bootloader is active when the L-LED (on D13) blinks after the upload start. So try to flash the original bootloader and look if the LED blinks after the upload is started. It can be found in the IDE directory under /hardware/arduino/avr/bootloaders/optiboot/optiboot_atmega328.hex

paulusjacobus, 2016-09-19:
Thanks that's a very smart approach that I did not think off. Will try it tonight, thumbs up.

paulusjacobus, 2016-09-20:
I have tried to upload the optiboot-atmega328.hex file but the led does not blink after the upload is started. 
After burning the bootloader, the led called L flashes a few times on the programmer UNO and the target UNO and then stops. I still cannot read the Id so I start to think my 328pb chip is damaged and will replace it with another one that I bought. I checked the solder pads and they are all fine. Will let you know how that goes.

awatterott, 2016-09-20:
Okay, then there seems to be a problem with the reset signal from the m16u2. After the upload is started the m328pb is resetted and the booloader starts (L led blinks).
What kind of board do you have and is it a 5V or 3,3V system?

paulusjacobus, 2016-09-20:
I believe the board is a 5V system. The voltage direct on the 328pb chip is 5v however there is a small breakout section of unused 3x4 pads next to the ICSP header with the following notation:
RX TX 5V GND
SCL SDA 5V GND
3.3V 3.3V GND GND

The pads with 3.3V do indeed have 3.3V.

paulusjacobus, 2016-09-22:
I have made some further progress. I loaded the 328.hex bootloader and when plugging in the usb cable directly, the L led flashes rapidly in random pattern and start than flashing in a sequence of 5 quick flashes with a pause. After a while the L led then turns on continuously. When uploading the blink sketch, the L led flashes and turns on when the IDE tries to sync.
Arduino: 1.6.11 (Windows 10), Board: "Atmel ATmega328PB Crystal Clock, 16 MHz"

D:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\Paul_2\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\Paul_2\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries C:\Users\Paul_2\Documents\Arduino\libraries -fqbn=m328pb:avr:atmega328pbcc:speed=16mhz -ide-version=10611 -build-path C:\Users\Paul_2\AppData\Local\Temp\build6755a1ce0f7e76a607d7c9aab3f4e3c5.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino5 -prefs=runtime.tools.avr-gcc.path=C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2 -verbose C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\examples\01.Basics\Blink\Blink.ino
D:\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino\hardware -hardware C:\Users\Paul_2\AppData\Local\Arduino15\packages -tools D:\Arduino\tools-builder -tools D:\Arduino\hardware\tools\avr -tools C:\Users\Paul_2\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\libraries -libraries C:\Users\Paul_2\Documents\Arduino\libraries -fqbn=m328pb:avr:atmega328pbcc:speed=16mhz -ide-version=10611 -build-path C:\Users\Paul_2\AppData\Local\Temp\build6755a1ce0f7e76a607d7c9aab3f4e3c5.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino5 -prefs=runtime.tools.avr-gcc.path=C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2 -verbose C:\Users\Paul_2\Downloads\arduino-nightly-windows\arduino-nightly\examples\01.Basics\Blink\Blink.ino
Using board 'atmega328pbcc' from platform in folder: C:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5
Using core 'arduino' from platform in folder: C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.13
Detecting libraries used...
"C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.13\cores\arduino" "-IC:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5\variants\atmega328pb" "C:\Users\Paul_2\AppData\Local\Temp\build6755a1ce0f7e76a607d7c9aab3f4e3c5.tmp\sketch\Blink.ino.cpp" -o "nul"
"C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.13\cores\arduino" "-IC:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5\variants\atmega328pb" "C:\Users\Paul_2\AppData\Local\Temp\build6755a1ce0f7e76a607d7c9aab3f4e3c5.tmp\sketch\Blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.13\cores\arduino" "-IC:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5\variants\atmega328pb" "C:\Users\Paul_2\AppData\Local\Temp\build6755a1ce0f7e76a607d7c9aab3f4e3c5.tmp\sketch\Blink.ino.cpp" -o "C:\Users\Paul_2\AppData\Local\Temp\build6755a1ce0f7e76a607d7c9aab3f4e3c5.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"D:\Arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Paul_2\AppData\Local\Temp\build6755a1ce0f7e76a607d7c9aab3f4e3c5.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Paul_2\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.13\cores\arduino" "-IC:\Users\Paul_2\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.0.5\variants\atmega328pb" "C:\Users\Paul_2\AppData\Local\Temp\build6755a1ce0f7e76a607 [... cut ...]
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 2 of 10: not in sync: resp=0x3d
avrdude.exe: ser_recv(): read error: Access is denied.

......

avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0x3d
the selected serial port avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0x3d
 does not exist or your board is not connected
avrdude.exe: ser_drain(): read error: Access is denied.

avrdude.exe done.  Thank you.

paulusjacobus, 2016-09-25:
Today I managed to borrow a digital scope to see whether the x-tal clock is working and it does, so the chip is active. I think next step is to verify the fuses, maybe this smd requires a slightly different selection?
The smd part number is mega328PB AU 1608

awatterott, 2016-09-25:
The fuse settings are the same. I think it has to do something with the reset signal of the m328pb.
If the crystal is not working then you also cannot flash/access the controller via ISP.

paulusjacobus, 2016-09-25:
Okay will have a look at the reset circuit and see whether I can improve things (i.e. try bigger capacitor o.i.d)

paulusjacobus, 2016-10-29:
Finally got it to work after lots of work and changes.
The AVR from the factory are set at 1MHz by default and don't program via the Arduino IDE or an Arduino as ISP setup.
I got a low cost ISP programmer USBasp and a program called eXtreme Burner which allows you to overwrite the factory set fuses.
After that you can program the AVR on a Uno R3 PCB via the Arduino IDE.

Note:
In order to get the USBasp to work you need the proper windows 10 drivers which you get from a tool called Zadig. This allows you to install the USB K driver.
You don't see a port on the Arduino IDE for the USBasp when burning the bootloader. I ended up using the extreme burner to program the optiboot.hex  file directly. After that I could use the Arduino IDE to program example sketches.

closed 2016-10-29 by paulusjacobus