watterott/ATmega328PB-Testing #12: package_m328pb_index.json not compatible with 1.6.16

markruys, 2017-01-09:
package_m328pb_index.json is not compatible with Arduino AVR Boards 1.6.16. You get compile errors like:
```
xxx:56: error: 'Serial' was not declared in this scope
```
Switch back to 1.6.15 in the board manager makes my code compile again.

awatterott, 2017-01-09:
I have tested it with IDE 1.8.0 + AVR Boards 1.6.16 and I get no errors.
Do you have replaced the files in the hardware folder?
If you install new tools (or boards which require new tools) to an old IDE then you have to replace the files in the user directory.

markruys, 2017-01-09:
H'm, I've must been sleeping, probably forgot to apply hardware.zip. IDE 1.8.1 and AVR boards 1.6.17 give me no troubles anymore. Sorry to have bothered you.

closed 2017-01-09 by markruys

ArduinoFanBoy, 2017-05-08:
I upgraded to IDE 1.8.2 and I can upload code to the 328PB, but I am not sure that SoftwareSerial works because it isn't recognizing a device I have (adafruit FONA 3G). Have you tested SoftwareSerial on the 328PB?

awatterott, 2017-05-08:
The SoftwareSerial lib works on the ATmega328PB. Please check the clock configuration (fuse bits) of the AVR microcontroller.

ArduinoFanBoy, 2017-05-08:
OK, thanks for the confirmation!