watterott/ATmega328PB-Testing #16: Error compiling

mizraith, 2017-02-01:
I'm attempting to compile the basic blink sketch (Arduino 1.8) and upload via the AVRISPmk2.  I'm getting the following error.     I'll take any suggestions/hints for getting it working.

`"/Applications/Arduino__181.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -mmcu=atmega328pb -DF_CPU=8000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/red/Library/Arduino15/packages/m328pb/hardware/avr/1.0.8/variants/atmega328pb" "/Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/var/folders/_0/khxnwnmm8xj15y0059_xt4th0000gn/T/arduino_build_89720/core/wiring_pulse.S.o"
In file included from /Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.S:52:0:
/Applications/Arduino__181.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^
"/Applications/Arduino__181.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega328pb -DF_CPU=8000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Users/red/Library/Arduino15/packages/m328pb/hardware/avr/1.0.8/variants/atmega328pb" "/Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/var/folders/_0/khxnwnmm8xj15y0059_xt4th0000gn/T/arduino_build_89720/core/WInterrupts.c.o"
/Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c: In function 'attachInterrupt':
/Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c:151:8: error: #error attachInterrupt not finished for this CPU (case 0)
       #error attachInterrupt not finished for this CPU (case 0)
        ^
/Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c: In function 'detachInterrupt':
/Applications/Arduino__181.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c:243:8: error: #error detachInterrupt not finished for this cpu
       #error detachInterrupt not finished for this cpu
        ^
exit status 1
Error compiling for board Atmel ATmega328PB Internal Clock.`

awatterott, 2017-02-01:
Have you done the [first installation step](https://github.com/watterott/ATmega328PB-Testing#installation-on-arduino-ide-1610) and added the files to the toolchain? Because the current Arduino toolchain does not support the ATmega328PB.

mizraith, 2017-02-01:
Thanks for the fast reply.  I suspected as much, but, I couldn’t quite figure out which files to update.

So, I’ve done some searching, but I don’t see arduino/hardware inside the Arduino application package.  
         I do see an old Arduino/hardware folder within my user directory, but it’s empty and appears unused.  

  I’m running on both mac and windows, so i’d have to do this change on 2 machines.   On Windows I see Arduino/hardware inside the Program Files(x86) directory.

I’ll take any hints you can throw my way…thanks!
_Red

mizraith, 2017-02-01:
Okay — that works for sure under Windows (updating the files under Program Files(x86)/Arduino/hardware/…

AHAH — I found the location on mac, it’s under:
   /Applications/Arduino/Contents/Java/hardware    

I think I got it from here!  Thanks for putting together the great ATMega328PB package.

_Red

mizraith, 2017-02-03:
## FIXING COMPILE ISSUES  
_(can you add these details to your Readme file)_

_In order to follow the first installation step_
**Windows:**
   Arduino source folder to update is under     `Program Files(x86)/Arduino/hardware`
**Mac**
   Arduino source folder to update is under     `/Applications/Arduino/Contents/Java/hardware`

**Issues with avrdude error part not found**
I still got an avrdude complaining  `avrdude: AVR Part "atmega328pb" not found.`  
Solution:
Copy the following and add it to your **avrdude.conf** file, found under source folder `/tools/avr/etc`

```
part parent "m328"
    id			= "m328pb";
    desc		= "ATmega328PB";
    signature		= 0x1e 0x95 0x16;

    ocdrev              = 1;
;

```
You can insert the above after the "ATmega328P" section.  For me it was around line 8506.

closed 2017-02-03 by mizraith

awatterott, 2017-02-03:
I have updated the installations step. But AVRdude should work, because the BSP includes an [avrdude.conf](https://github.com/watterott/ATmega328PB-Testing/blob/master/hardware/atmega328pb/avr/tools/avrdude.conf) with the 328PB definition and this file is used by the [platform.txt](https://github.com/watterott/ATmega328PB-Testing/blob/master/hardware/atmega328pb/avr/platform.txt#L93)

soulsbysynths, 2018-04-08:
I've just had an identical issue to above, however I did the install with both .json URLs added to preferences as per instructions.
Solution was to downgrade the Arduino AVR Boards (in boards manager) to 1.6.200 (currently on 1.6.21) and then all was fine.

curie31, 2020-03-24:
> Have you done the [first installation step](https://github.com/watterott/ATmega328PB-Testing#installation-on-arduino-ide-1610) and added the files to the toolchain? Because the current Arduino toolchain does not support the ATmega328PB.

Hello, i am following the steps for the installation. I am getting the following error,
 Error downloading https://downloads.arduino.cc/packages/package_index.json

Also, i see a warning in the verbose output,
WARNING: Error loading hardware folder C:\Users\hp\Documents\Arduino\hardware\ATmega328PB-Testing-master

I am trying to code my ATmega328PB board using Arduino ISP as programmer. Avrdude says,
Part m328pb not found.

Please help!!

awatterott, 2020-03-24:
Please follow the installations steps and use the Boards Manager for the installation:
https://github.com/watterott/ATmega328PB-Testing#installation-on-arduino-ide-18

curie31, 2020-03-25:
> Please follow the installations steps and use the Boards Manager for the installation:
> https://github.com/watterott/ATmega328PB-Testing#installation-on-arduino-ide-18

Thank you for the quick reply.