watterott/ATmega328PB-Testing #17: Will you post the bootloader's source code?

liudr, 2017-03-23:
Hi,

Thanks for the project! I am interested in using the second serial port but encountered some strange problem. I use Serial1 for a GPS module. Somehow there is some issue with RX1 not receiving anything although the pin receives information (checked with an FTDI adapter on the pin). I wonder if there's anything in the bootloader that would cause problems.

Also, if I connect the GPS to the RX1 and TX1, since they are also SPI MISO and MOSI, will the GPS interfere with program uploading?

awatterott, 2017-03-25:
Hi,
I think the bootloader can not be a problem.
The bootloader is Optiboot with ATmega328PB signature and the source code is in the wattuino repo.

liudr, 2017-03-25:
Thank you Andreas. I figured our my problem. It wasn't related to ATMEGA328PB. It was EMI from the GPS module. It only got severe enough when the GPS module got too close to the board with 328PB. I just figured it out. Here is a link to the post:

http://forum.arduino.cc/index.php?topic=465110.0

closed 2017-03-25 by liudr