watterott/ATmega328PB-Testing #18: spi not working

goldkwang, 2017-04-06:
arduino uno (Atmega328P) -> spi working
Atmega328PB - spi not working

i test nrf8001 bluetooth module (spi)

awatterott, 2017-04-07:
The m328PB has two SPI ports (libraries SPI.h and SPI1.h).
Have you checked the signals with an oscilloscope or logic analyzer?

goldkwang, 2017-04-07:
I checked the signal using oscilloscope.
SPI & SPI1 signal is not displayed. ( use watterott )
So I edit the arduino uno source and download hex to Atmega328PB -> good working

My development environment :
Atmega328PB, 16MHz, 5V
SPI Connect -> nRF8001 Bluetooth Module
Atmel Studio7 (7.0.1417)
Arduino IDE 1.8.3
AVRJTACICE MKII & AVRISP MKII

awatterott, 2017-04-08:
What do you mean with "SPI & SPI1 signal is not displayed"?
Do you use the Arduino IDE or Atmel Studio?

goldkwang, 2017-04-09:
i mean it's not displayed at oscilloscope.
Source Compile -> Arduino IDE
Firmware Download -> Atmel Studio

awatterott, 2017-04-09:
Have you tried a basic blink program to verify that the building/compiling is working?
I have updated the SPI libs so that the SPI and SPI1 code is now the same (except registers).

goldkwang, 2017-04-10:
Arduino basic blink Example -> Good Working
PortE Blink & PortE Input Button -> Good Working

Current state changed communication (SPI -> UART)
UART0, UART1, PortE Input, PortE Output , Analog Input (A0) -> Good Working
I did not test the SPI by removing the SPI module from the circuit.

awatterott, 2017-04-12:
Have you tried the new BSP with updated SPI libs?

awatterott, 2017-04-21:
Closing because of inactivity.

closed 2017-04-21 by awatterott