watterott/ATmega328PB-Testing #19: Can't get it to work

botletics, 2017-05-04:
Hi there, could someone please explain step-by-step how to use this repo? I know people have used this and gotten it to work, but I can't seem to. I added the 328PB boards using Boards Manager but I'm a bit confused as to which files I need to update under /arduino/hardware. Also, exactly how do I load the code onto the 328PB chip? (For example, do I just select the ATmega328PB with Crystal Clock/16 MHz and upload the sketch?)

Originally what I did before coming here is I loaded ArduinoISP into an Arduino Uno and used it to burn the standard 328P bootloader into the 328PB chip, and that ran successfully. In order for that to work I changed the avrdude.conf file device signature for the ATmega328P, then changed it back after I flashed the bootloader. However, I can't seem to upload sketches onto the 328PB chip with my FTDI, even though it has the bootloader. Do I need a special 328PB bootloader in order to upload sketches to the chip?

Additional info: I am using an ATmega328PB chip with a 10k pullup resistor on the RESET pin and 16MHz resonator.

Is there something I'm missing here? Thanks!

ArduinoFanBoy, 2017-05-08:
Update: I can now successfully upload the bootloader as well as code with the Arduino as ISP and with the "Upload Using Programmer" option in the IDE. However, I have to select 1MHz Internal for the 328PB even though I have a 16MHz resonator, because all other clock settings have the wrong timing (for example, 1s delay would be more like 15s or something). Any ideas why the timing isn't correct?

Also, I would still love to be able to upload code via an FTDI so that I can use the serial monitor without disconnecting the Arduino Uno then plugging in the FTDI each time I change the code. Right now, even with the bootloader successfully flashed I cannot upload code with FTDI.

awatterott, 2017-05-08:
I think you have not set the fuse bits and so the AVR runs with the default settings: internal 8MHz RC oscillator with 1/8 prescaler. With 1MHz clock the bootloader will also not work, because the default UART speed (115200) is too high.
Try to flash the [bootloader](https://github.com/watterott/wattuino/raw/master/software/Optiboot/optiboot_m328pb.hex) directly with AVRdude. Here are the settings:
https://github.com/watterott/wattuino/tree/master/software/Optiboot#fuse-settings-for-atmega328

ArduinoFanBoy, 2017-05-08:
It seems like the default speed is actually 1MHz because all the other options yield the wrong timing. I can successfully burn the bootloader with the IDE at 16MHz but it seems as if it is still running at 1MHz because a 1s delay becomes 16s, so somehow it's not setting the fuses properly I guess? I also can't upload code with the FTDI after the bootloader is flashed.

awatterott, 2017-05-08:
The fuse settings of m328, m328p and m328pb are the same and you have to use the right [avrdude.conf](https://github.com/watterott/ATmega328PB-Testing/blob/master/hardware/atmega328pb/avr/tools/avrdude.conf).
Please try it with the Arduino IDE and enable the verbose output on upload and send the log messages, when burning the bootloader.

ArduinoFanBoy, 2017-05-08:
Here is what I chose:

[image]

Here is the verbose output:

avrdude.exe: Version 6.3, compiled on Jan 17 2017 at 12:00:53
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\atmega328pb\avr/tools/avrdude.conf"

             Using Port                    : COM7
             Using Programmer              : stk500v1
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega328PB
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9516 (probably m328pb)
avrdude.exe: erasing chip
avrdude.exe: reading input file "0xFF"
avrdude.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 1 bytes of lock written
avrdude.exe: verifying lock memory against 0xFF:
avrdude.exe: load data lock data from input file 0xFF:
avrdude.exe: input file 0xFF contains 1 bytes
avrdude.exe: reading on-chip lock data:
[... 121 lines cut ...]
avrdude.exe: load data flash data from input file C:\Program Files (x86)\Arduino\hardware\atmega328pb\avr/bootloaders/optiboot_m328pb.hex:
avrdude.exe: input file C:\Program Files (x86)\Arduino\hardware\atmega328pb\avr/bootloaders/optiboot_m328pb.hex contains 32748 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude.exe: verifying ...
avrdude.exe: 32748 bytes of flash verified
avrdude.exe: reading input file "0xCF"
avrdude.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude.exe: 1 bytes of lock written
avrdude.exe: verifying lock memory against 0xCF:
avrdude.exe: load data lock data from input file 0xCF:
avrdude.exe: input file 0xCF contains 1 bytes
avrdude.exe: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude.exe: verifying ...
avrdude.exe: 1 bytes of lock verified

avrdude.exe done.  Thank you.

awatterott, 2017-05-08:
You have to choose as board "Atmel ATmega328PB Crystal Clock" otherwise the fuses are set for internal clock.

ArduinoFanBoy, 2017-05-08:
Yes, for now I want to use 8MHz internal clock. I also tried 16MHz Crystal Clock and it burns the bootloader just fine, but when I upload code to it, 1s delay becomes 16s delay. Likewise with the 8MHz, 1s becomes 8s delay. It seems like the bootloader isn't actually setting it to the right clock rate even though it seems successful.

awatterott, 2017-05-08:
Please send the log messages to confirm the fuses, otherwise I cannot help.

ArduinoFanBoy, 2017-05-08:
Sorry, where are the log messages? Thanks.

awatterott, 2017-05-08:
I mean the log messages, when burning the bootloader with the correctly selected board.

ArduinoFanBoy, 2017-05-08:
I'm sorry, I'm still not quite following what log message you are referring to. I pasted the entire verbose output (the stuff that prints out at the bottom).

ArduinoFanBoy, 2017-05-08:
Also, one tweak I had to make using this Github repo is I had to change the avrdude.conf file located in C:\Program Files (x86)\Arduino\hardware\tools\avr\etc to be able to upload the code with "Upload Using Programmer". I added the following code at line 8711:

part parent "m328"
    id      = "m328pb";
    desc    = "ATmega328PB";
    signature   = 0x1e 0x95 0x16;

    ocdrev              = 1;
;

Without that tweak, it would give me the message:

System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"
avrdude: AVR Part "atmega328pb" not found.

Would my tweak cause any issues?

awatterott, 2017-05-08:
If you select as board "Atmel ATmega328PB Internal Clock" then the fuses will be set to use the internal clock and no external crystal, when burning the bootloader. The right board is "Atmel ATmega328PB Crystal Clock" for an external crystal and with this settings you should send the log messages.

ArduinoFanBoy, 2017-05-08:
For now I would like to use 8MHz Internal, not the crystal. However, I ran it with 16MHz and here is the verbose output:

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude.exe -CC:\Program Files (x86)\Arduino\hardware\atmega328pb\avr/tools/avrdude.conf -v -patmega328pb -cstk500v1 -PCOM7 -b19200 -e -Ulock:w:0xFF:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude.exe: Version 6.3, compiled on Jan 17 2017 at 12:00:53
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\atmega328pb\avr/tools/avrdude.conf"

             Using Port                    : COM7
             Using Programmer              : stk500v1
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega328PB
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9516 (probably m328pb)
avrdude.exe: erasing chip
avrdude.exe: reading input file "0xFF"
avrdude.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 1 bytes of lock written
avrdude.exe: verifying lock memory against 0xFF:
avrdude.exe: load data lock data from input file 0xFF:
avrdude.exe: input file 0xFF contains 1 bytes
avrdude.exe: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s
[... 119 lines cut ...]
avrdude.exe: load data flash data from input file C:\Program Files (x86)\Arduino\hardware\atmega328pb\avr/bootloaders/optiboot_m328pb.hex:
avrdude.exe: input file C:\Program Files (x86)\Arduino\hardware\atmega328pb\avr/bootloaders/optiboot_m328pb.hex contains 32748 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% -0.00s

avrdude.exe: verifying ...
avrdude.exe: 32748 bytes of flash verified
avrdude.exe: reading input file "0xCF"
avrdude.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.03s

avrdude.exe: 1 bytes of lock written
avrdude.exe: verifying lock memory against 0xCF:
avrdude.exe: load data lock data from input file 0xCF:
avrdude.exe: input file 0xCF contains 1 bytes
avrdude.exe: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude.exe: verifying ...
avrdude.exe: 1 bytes of lock verified

avrdude.exe done.  Thank you.

awatterott, 2017-05-08:
The messages do not look okay. Maybe because of the Markdown parser.
Please put them code tags.

ArduinoFanBoy, 2017-05-08:
I have attached a .txt file with the verbose output. I think that should be formatted better, sorry.
[VerboseOutput_8MHzInternal.txt](https://github.com/watterott/ATmega328PB-Testing/files/984405/VerboseOutput_8MHzInternal.txt)

awatterott, 2017-05-08:
Its the same. Have a look at line 94 - there is an avrdude start, but the previous start is not finished.
Please use also an external crystal, because the internal clock is not very accurate and so there can be problems with the UART.

ArduinoFanBoy, 2017-05-08:
I have a 16MHz resonator, but I can't use it at 3.3V so that's why I am switching to 8MHz internal.

awatterott, 2017-05-08:
Have you tried it? With a normal temperature of about 20°C there will be no problems.

ArduinoFanBoy, 2017-05-08:
Yes, when I upload blink.ino at 16MHz it again reverts back to 1MHz internal for some reason. (1s delay becomes 16s). Same with 8MHz internal.

awatterott, 2017-05-08:
This can also happen, when you select the wrong board/clock settings in the Arduino IDE.
Burn the bootloader with the correct board (external crystal) and then try to flash a Sketch via the bootloader.

ArduinoFanBoy, 2017-05-08:
Yes, I tried that. I burned the bootloader at 16MHz then used "Upload Using Programmer" with my Arduino as ISP and the timing was off. I could never seem to upload via FTDI with the bootloader in place. I can only upload via "Upload Using Programmer" from the IDE while connecting the chip to the Arduino via SPI.

ArduinoFanBoy, 2017-05-08:
I found out that 8MHz now works (I had to manually reset the chip otherwise it would blink every 8s instead of 1s). 16MHz doesn't work per the spec sheet.

closed 2017-05-08 by ArduinoFanBoy