watterott/ATmega328PB-Testing #21: SPI not working correctly with V1.0.9

Yoshi20, 2017-05-12:
Just to let you know:

I'm using Arduino 1.6.12 and I'm controlling an E-Paper-Display with SPI1.
Since V1.0.9 it doesn't work any more (does nothing I guess).
V1.0.8 however works fine.

I tried V1.0.9 with Arduino_1.8.2 but facing the following problem when compiling:
/Applications/Arduino_1.8.2.app/Contents/Java/hardware/tools/avr/avr/include/avr/sleep.h:224:6: error: #error "No SLEEP mode defined for this device."

Thanks for your great work.

awatterott, 2017-05-12:
I have updated the SPI libs with v1.0.9 and I know see that there is a problem when using SPI+SPI1.
Can you try v1.10.
On the other error: Have you added/updated all files from the hardware.zip archive? It looks like there is something missing.

Yoshi20, 2017-05-16:
I now tested V1.1.0 but no improvements!
Only V1.0.8 works to control the EPD with SPI1.

Concerning the second bug:
I now updated the Hardware files and after adding the m328pb in the avrdude.conf-file (what you by the way do not mention in the installing guide) I'm now able to use the ATmega328PB boards in Arduino_1.8.2.

In the avrdude.conf-file I added on line number 8506:
```
part parent "m328"
    id      = "m328pb";
    desc    = "ATmega328PB";
    signature    = 0x1e 0x95 0x16;

    ocdrev              = 1;
;
```

awatterott, 2017-05-16:
I made some tests and SPI and SPI1 is working for me.
Does your display work with SPI?

The correct [avrdude.conf](https://github.com/watterott/ATmega328PB-Testing/tree/master/hardware/atmega328pb/avr/tools) with m328PB definition is included with the boards package and used for the "Atmel ATmega328PB..." boards (via [boards.txt](https://github.com/watterott/ATmega328PB-Testing/blob/master/hardware/atmega328pb/avr/platform.txt#L93)).
Have you done all 3 [installation steps](https://github.com/watterott/ATmega328PB-Testing#installation-on-arduino-ide-18)?

Yoshi20, 2017-05-17:
Yes and Yes.

I now found the time to test the two different versions (1.1.0 and 1.0.8) and discovered which line is the problem:
In V1.1.0 -> SPI1.cpp -> LineNr: 74
`SPCR1 &= ~_BV(SPE);`
If replaced with:
`SPCR0 &= ~_BV(SPE);  // like in V1.0.8`
it works again :)

I'm looking forward to V1.1.1

awatterott, 2017-05-17:
The line in v1.0.8 is not correct, because it deactivates the wrong SPI interface.
Have you tested your display with both SPI interfaces (SPI/SPI0 and SPI1)?

Yoshi20, 2017-05-17:
No, it is on a PCB and I can only use SPI1.
I'm using the following pinout:
PC1 -> SCK1
PC0 -> MISO1
PE3 -> MOSI1
PE2 -> CS1 (SS1)

If it was wrong in V1.0.8, then I wonder why it only works with this one.

awatterott, 2017-05-17:
Maybe there is an error in your code. Do you use the SPI1.end() function?

Yoshi20, 2017-05-17:
You seem to be right in the end :)

I now reviewed the EPD-library I'm using and discovered, that it calls SPI1.end() way too early (after a draw and not only in the EPD.end()-function).
I fixed this and it now seems to work fine with V1.1.0.

Thanks for your help.

closed 2017-05-17 by Yoshi20