watterott/ATmega328PB-Testing #23: NRF24L01 SPI not working

javjongens, 2017-06-07:
Hello Andreas,

currently I am working on project with an NRF24L01 transceiver module on an atmega328pb. Everything is wired up correctly and has power (atmega328pb 5V, 16Mhz crystal | NRF24L01 3,3V, bypass caps included) - but i cant get the SPI communication between the nrf24l01 module and the atmega328pb working. The NRF24L01 module is connected to SPI0 (Arduino Pins 11,12,13).

If I change the atmega328pb to an atmega328p everything works well.

Do you have an idea where the problem might be?

Thanks,

jav

awatterott, 2017-06-08:
Have you changed the microcontroller or do you have selected another board/controller in the Arduino IDE?

javjongens, 2017-06-08:
Hello Andreas,

thanks for your response. Let me explain my setup:

Setup #1:
- Breadboard
- Arduino Nano (atmega328p)
- NRF24L01 Module via SPI (3,3V Supply via 3,3V LDO, 10uF + 100nF Cap soldered directly to the NRF24L01 Module, tmrh20 Library).
--> Scanner Example (to check basic functionality) works, eg shows me activity on 2,4 GHz Band due to wifi

Setup #2:
- own PCB (double checked all traces)
- ATmega328pb (with your Arduino IDE modification, correct board + crystal frequency selected in Arduino IDE)
- Same NRF24L01 Module from Setup #1.
--> Scanner Example not working, printDetails() (read all registers in the NRF24L01 chip) shows different values compared to setup #1

Do you have any idea?

Best regards,

jav

awatterott, 2017-06-08:
Are the fuse bits correctly set on the m328PB?
The m328PB is software compatible with the m328P. So you can upload the working m328P program/sketch to the m328PB. If this works, then it is a software problem (in the core or lib) and otherwise a hardware problem.

awatterott, 2017-06-18:
@javjongens Are there any news?

javjongens, 2017-06-18:
Dear Andreas,

I'am quite ashamed to tell you that I found the error. Of course it was not in your library. It was the series resistors inside the spi lines that messed up the signals. :/

Sorry that i stole your time and thanks for your great library.

Best regards,

jav

awatterott, 2017-06-18:
No problem.

closed 2017-06-18 by awatterott