watterott/ATmega328PB-Testing #24: "Device type not defined" and "interrupt attach/detach not completed"

Wor3d, 2017-07-26:
Hi, just installed fresh watterott files for the 328pb (got my own board with 328pb), every step in the manual was done in order and I am getting these errors. I am using USBasp as a programmer, bootloader was burnt fine (no errors).

````
C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino\WInterrupts.c: In function 'attachInterrupt':

C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino\WInterrupts.c:151:8: error: #error attachInterrupt not finished for this CPU (case 0)

       #error attachInterrupt not finished for this CPU (case 0)

        ^

C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino\WInterrupts.c: In function 'detachInterrupt':

C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino\WInterrupts.c:243:8: error: #error detachInterrupt not finished for this cpu

       #error detachInterrupt not finished for this cpu

        ^
exit status 1
Nastala chyba pri kompilácii pre dosku Atmel ATmega328PB Crystal Clock.`
```

Full upload message:
```
D:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Program Files\Arduino\hardware -hardware C:\Users\Notebook\AppData\Local\Arduino15\packages -tools D:\Program Files\Arduino\tools-builder -tools D:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Notebook\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files\Arduino\libraries -libraries C:\Users\Notebook\Documents\Arduino\libraries -fqbn=m328pb:avr:atmega328pbcc:speed=16mhz -ide-version=10803 -build-path C:\Users\Notebook\AppData\Local\Temp\arduino_build_78194 -warnings=none -build-cache C:\Users\Notebook\AppData\Local\Temp\arduino_cache_797876 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\Notebook\Documents\Arduino\ThermostatV3\Program\Program.ino
D:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware D:\Program Files\Arduino\hardware -hardware C:\Users\Notebook\AppData\Local\Arduino15\packages -tools D:\Program Files\Arduino\tools-builder -tools D:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\Notebook\AppData\Local\Arduino15\packages -built-in-libraries D:\Program Files\Arduino\libraries -libraries C:\Users\Notebook\Documents\Arduino\libraries -fqbn=m328pb:avr:atmega328pbcc:speed=16mhz -ide-version=10803 -build-path C:\Users\Notebook\AppData\Local\Temp\arduino_build_78194 -warnings=none -build-cache C:\Users\Notebook\AppData\Local\Temp\arduino_cache_797876 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\Notebook\Documents\Arduino\ThermostatV3\Program\Program.ino
Using board 'atmega328pbcc' from platform in folder: C:\Users\Notebook\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.0
Using core 'arduino' from platform in folder: C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19
Detecting libraries used...
"C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.0\variants\atmega328pb" "C:\Users\Notebook\AppData\Local\Temp\arduino_build_78194\sketch\Program.ino.cpp" -o "nul"
"C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.0\variants\atmega328pb" "-IC:\Users\Notebook\Documents\Arduino\libraries\Adafruit_NeoPixel" "C:\Users\Notebook\AppData\Local\Temp\arduino_build_78194\sketch\Program.ino.cpp" -o "nul"
"C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.0\variants\atmega328pb" "-IC:\Users\Notebook\Documents\Arduino\libraries\Adafruit_NeoPixel" "-IC:\Users\Notebook\Documents\Arduino\libraries\OneWire-master" "C:\Users\Notebook\AppData\Local\Temp\arduino_build_78194\sketch\Program.ino.cpp" -o "nul"
"C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.0\variants\atmega328pb" "-IC:\Users\Notebook\Documents\Arduino\libraries\Adafruit_NeoPixel" "-IC:\Users\Notebook\Documents\Arduino\libraries\OneWire-master" "-IC:\Users\Notebook\Documents\Arduino\libraries\dallas-temperature-control" "C:\Users\Notebook\AppData\Local\Temp\arduino_build_78194\sketch\Program.ino.cpp" -o "nul"
"C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.0\variants\atmega328pb" "-IC:\Users\Notebook\Documents\Arduino\libraries\Adafruit_NeoPixel" "-IC:\Users\Notebook\Documents\Arduino\libraries\OneWire-master" "-IC:\Users\Notebook\Documents\Arduino\libraries\dallas-temperature-control" "C:\Users\Notebook\Documents\Arduino\libraries\Adafruit_NeoPixel\Adafruit_NeoPixel.cpp" -o "nul"
"C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10803 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.19\cores\arduino" "-IC:\Users\Notebook\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.0\variants\atmega328pb" "-IC:\Users\Notebook\Documents\Arduino\libraries\Adafruit_NeoPixel" "-IC:\Users\Notebook\Documents\Arduino\libraries\OneWire-master" "-IC:\Users\Notebook\Documents\Arduino\libraries\dallas-temperature-control" "C:\Users\Notebook\Documents\Arduino\libraries\Adafruit_NeoPixel\esp8266.c" -o "nul"
"C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sectio [... cut ...]
       #error detachInterrupt not finished for this cpu

        ^

Používam knižnicu Adafruit_NeoPixel vo verzii 1.0.3 v adresáry: C:\Users\Notebook\Documents\Arduino\libraries\Adafruit_NeoPixel 
Používam knižnicu OneWire-master vo verzii 2.3.3 v adresáry: C:\Users\Notebook\Documents\Arduino\libraries\OneWire-master 
Použitá knižnica dallas-temperature-control v adresáry: C:\Users\Notebook\Documents\Arduino\libraries\dallas-temperature-control (legacy)
exit status 1
Nastala chyba pri kompilácii pre dosku Atmel ATmega328PB Crystal Clock.

```

awatterott, 2017-07-26:
It looks like you are using an older IDE and that you have installed a new core + toolchain. In this case you have to update/replace the files in the user directory:`"C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\`
Or you can also use the current IDE v1.8.3, which includes the latest toolchain.

Wor3d, 2017-07-26:
I am using the v1.8.3 :) So did I actually harm something by installing the toolchain manually? I haven't seen "do not install toolchain if you have IDE vx.x.x or above" anywhere...

awatterott, 2017-07-26:
If you use v1.8.3 then the AVR Boards v1.6.19 (with the new toolchain) cannot be installed or uninstalled, because they are bundled with the IDE.

Wor3d, 2017-07-27:
I am talking about installing the Atmega328pb toolchain (sorry if I have chosen my words poorly...)
When I am using 328p-pu, everything is compiled OK, but I cannot upload 328p-pu program on PB without modifying the checksum etc...

awatterott, 2017-07-27:
There is no special ATmega328PB toolchain and the current Arduino toolchain does not include the files/definitions for the ATmega328PB.
If you want to upload a 328P program to a 328PB, then you can modify the AVRdude configuration (avrdude.conf) and change the signature.
I would recommend to start with a clean installation of the Arduino IDE and remove the packages/directories under `C:\Users\Notebook\AppData\Local\Arduino15\packages\arduino\`.
Then the [installation steps](https://github.com/watterott/ATmega328PB-Testing#installation-on-arduino-ide-18) will work as long as no other "AVR Boards" package is installed via the Boards Manager.

closed 2017-07-27 by awatterott

Wor3d, 2017-07-27:
Well, what is your software if not 328pb toolchain? :)

I will re-install IDE and delete the folder as you've said and give it a go on a clean IDE :)

Wor3d, 2017-07-29:
Well, here I am again, deleted the folder and re-installed the Arduino IDE, still got the same problem...

awatterott, 2017-07-30:
Pleased post the log messages from the compiler.

awatterott, 2017-08-02:
The next Arduino AVR toolchain will have support for the ATmega328PB:
https://github.com/arduino/toolchain-avr/pull/47