watterott/ATmega328PB-Testing #25: Serial not defined

szwetsloot, 2017-12-02:
When trying to use these board definitions with the arduino IDE I get:
Main:106: error: 'Serial' was not declared in this scope
Serial.begin(9600);

Code:
`// Written on         12/11/2017
// Last update        12/11/2017
// Author             Sybren Zwetsloot
// Current version    1.0
// Version log
// 1.0                First version - Created for the new PCB for the screen.

// Pins
#define     LR        8       // Red background LED
#define     LG        9       // Green background LED
#define     LB        10      // Blue background LED
#define     SID       7       // Screen  Serial data output (SID)
#define     SCLK      6       // Screen - Serial clock output (SCLK)
#define     A0_       5       // Screen - Data / Command select (A0)
#define     RST       4       // Screen - RESET (RST)
#define     CS        3       // Screen - LCD chip select (CS)
//#define     SDA       A4      // I2C data line
//#define     SCL       A5      // I2C clock line
#define     CHARGE    A7      // High when charging
#define     BATTERY   0       // Battery voltage input

// Screen
#define     POSITIVE  0       // Defines whether the screen is positive or negative
#if POSITIVE
#define contrastdef 0x19
#else
#define contrastdef 0x1F
#endif

// Timing
#define SCREEN_REFRESH_PERIOD   500                       // ms
#define ACCEL_REFRESH_FREQ      100                       // HZ
#define ACCEL_DT                1 / ACCEL_REFRESH_FREQ    // s

#include "Arduino.h"
#include <Wire.h>
#include "Fonts.h"
#include "Icons.h"
#include "Structures.h"
#include "Accel.h"
#include "Screen.h"
// TEMP
//#include "ST7565.h"

// TEMP
//ST7565 glcd(_SID, _SCLK, _A0, _RST, _CS);
//#define LOGO16_GLCD_HEIGHT 16
//#define LOGO16_GLCD_WIDTH  16

// Variables
// Classes
screen glcd;

// Structures
unit_t unit;
timing_t timing;
accel_t accel;

// Prototypes
void measureBattery();

void setup() {
  accel.lastAngle.x = 1;
  accel.lastAngle.y = 0;
  accel.lastAngle.z = 0;
  accel.previousLowPassValue = 0.0;
  accel.previousHighPassValue = 0.0;
  accel.toggle = 0;
  accel.max = 0;
  accel.first = 1;
  
  // Pin modes
  pinMode(LR, OUTPUT);
  pinMode(LG, OUTPUT);
  pinMode(LB, OUTPUT);

  pinMode(SCL, OUTPUT);
  pinMode(SDA, OUTPUT);

  pinMode(CHARGE, INPUT);
  //pinMode(BATTERY, INPUT);

  // Default digital writes
  digitalWrite(LR, LOW);
  digitalWrite(LG, LOW);
  digitalWrite(LB, LOW);

  /*glcd.begin(0x1F);
  glcd.clear();
  accelInit();
  while(1) {
    if (millis() > base.lastAccelRead) {
      accelRead(accel);
      calculateStrokerate(accel);
      base.lastAccelRead += 10;
    }
  }*/
  

  glcd.init();
  glcd.setBrightness(contrastdef);
  glcd.drawLogo();
  delay(1000);

  // Initialize serial connection
  Serial.begin(9600);

  // Set the voltage reference
  analogReference(INTERNAL);
  // Do some measurements to stabilize
  for (uint8_t i = 0; i < 255; i++) {
    analogRead(BATTERY);
  }
  delay(10);

  // TEMP - Initialize the screen
  if (!accelInit()) {
    // TODO - draw an error screen
    while (1);
  }

  // Set the default variable states
  unit.gpsState = 0;
  unit.split = 0;
  unit.strokeRate = 0;
  unit.time = 0;
  unit.distance = 0;
  measureBattery();
  glcd.draw(unit);
  timing.lastAccelRead = millis();
  timing.lastBatteryMeasurement = millis();
  timing.lastScreenRefresh = millis();
}

int16_t ax, ay, az;
void loop() {
  // Check whether the charger is connected
  if (digitalRead(CHARGE)) {
    if (unit.state == 1) {
      unit.state = 0;
    }
  } else {
    if (unit.state == 0) {
      unit.state = 1;
    }
  }

  // Check the accelerometer
  if (millis() > timing.lastAccelRead) {
    timing.lastAccelRead += 10;
    accelRead(accel);
    calculateStrokerate(accel);
  }

  // Check whether we need to draw the screen
  if (millis() > timing.lastScreenRefresh) {
    if (millis() - accel.lastStroke > 6000) {
      unit.timeOffset += SCREEN_REFRESH_PERIOD;
      unit.fillCount = 0;
      unit.strokeCount = 3;
      unit.velocityCount = 3;
    }
    timing.lastScreenRefresh += SCREEN_REFRESH_PERIOD;
    if (unit.gpsState == 1) {
      if (millis() - unit.lastDistanceUpdate < 4000) {
        unit.drawnDistance = unit.distance + (uint32_t)(round(float(arrayMean(unit.velocityLog))) * (millis() - unit.lastDistanceUpdate) / 1000);
      }
    }
    if (unit.state == 0) {
      unit.velocity = arrayMean(unit.velocityLog);
      if (unit.velocity > (500.0 / 599.0)) { // Make sure we're not dividing by zero
        unit.split = (uint8_t)round(500.0 / unit.velocity);
      } else {
        unit.split = 0;
      }
      unit.strokeRate = arrayMean(unit.strokeLog);
      glcd.draw(unit);
    } else if (unit.state == 1) {
      glcd.drawCharging();
    }
  }

  // Measure the battery voltage
  if (millis() > timing.lastBatteryMeasurement) {
    timing.lastBatteryMeasurement += 100;
    measureBattery();
  }
  delay(1);
}

void measureBattery() {
  // Set the voltage reference
  analogReference(INTERNAL);
  // Do some measurements to stabilize
  for (uint8_t i = 0; i < 8; i++) {
    analogRead(BATTERY);
  }
  delay(5);
  int val = 0;
  for (uint8_t i = 0; i < 20; i++){
    val += analogRead(BATTERY);
    delayMicroseconds(100);
  }
  val /= 20;
  if (val < 712) val = 712;
  if (val > 907) val = 907;
  unit.batteryPercentage = (uint8_t) ((((float)val) - 712.0) / 195.0 * 100.0);
}

uint16_t arrayMean(uint16_t *arr) {
  if (unit.fillCount >= 3) {
    return uint16_t round(float(arr[0] + arr[1] + arr[2]) / 3);
  } else {
    if (unit.fillCount == 0)  return 0;
    uint32_t ret = 0;
    for (uint8_t i = 0; i < unit.fillCount; i++)
      ret += arr[i];
    return uint16_t(ret / unit.fillCount);
  }
}
float arrayMean(float *arr) {
  if (unit.fillCount >= 3) {
    return (arr[0] + arr[1] + arr[2]) / 3;
  }  else {
    if (unit.fillCount == 0)  return 0.0f;
    float ret = 0.0f;
    for (uint8_t i = 0; i < unit.fillCount; i++)
      ret += arr[i];
    return ret / unit.fillCount;
  }
}
uint8_t arrayMean(uint8_t *arr) {
  if (unit.fillCount >= 3) {
    return uint8_t round(float(arr[0] + arr[1] + arr[2]) / 3);
  } else {
    if (unit.fillCount == 0)  return 0;
    uint32_t ret = 0;
    for (uint8_t i = 0; i < unit.fillCount; i++)
      ret += arr[i];
    return uint8_t(ret / unit.fillCount);
  }
}`

awatterott, 2017-12-02:
Which IDE and AVR core version do you use?

szwetsloot, 2017-12-02:
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
Arduino IDE 1.8.5

awatterott, 2017-12-02:
Do you have installed the AVR core 1.6.200? The built-in core has no support for the ATmega328PB.
https://downloads.arduino.cc/packages/package_avr_3.6.0_index.json

szwetsloot, 2017-12-02:
Does that include either this one:
[image]
Or this one:
[image]
?

If not, can youtell me how to install it?

awatterott, 2017-12-02:
Add the [link](https://downloads.arduino.cc/packages/package_avr_3.6.0_index.json) under File->Preferences and then install the board with Tools->Board->Boards Manager

szwetsloot, 2017-12-02:
Sorry for the stupid questions, which board do you mean?

szwetsloot, 2017-12-02:
Is Arduino AVR Board 1.6.200 the same as Arduino AVR Board 1.6.20?
I can't uninstall and reinstall that package.
I tried to downgrade to Arduino AVR Board 1.6.19 and upgrade to Arduino AVR Board 1.6.20 again, but that did not change anything.

awatterott, 2017-12-02:
No, 1.6.20 is not 1.6.200. You have to add the link before you can install the new board.

szwetsloot, 2017-12-02:
I though I did, but apparently misclicked.
That resolved my issue.
Thank you for the quick help!

closed 2017-12-02 by szwetsloot