watterott/ATmega328PB-Testing #31: Sleep and wake up on ext. interrupt not working

leozulfiu, 2018-03-21:
### The Problem

I would like to put the atmega328pb to sleep and wake up again when pressing a button which is connected to an external interrupt pin. I tried it with the attached sketch but it doesn't work really stable.

**Observations**
- When I press the button during the counting up (awake phase) the variable `count` is increased with each button push. IMO pressing the button shouldn't affect the loop execution at all.
- Holding down the button during the loop execution and following sleep phase results in a microcontroller deadlock. It becomes completly unresponsive and a new upload of the sketch is needed.

### Example Code

```
#include <avr/sleep.h>

int wakePin = 2;
int sleepStatus = 0;
int count = 0;
 
void wakeUpNow() {
  detachInterrupt(0);
}
 
void setup() {
  pinMode(wakePin, INPUT);
  Serial.begin(9600);
  attachInterrupt(0, wakeUpNow, LOW);
}
 
void sleepNow() {
    set_sleep_mode(SLEEP_MODE_PWR_DOWN);
    sleep_enable();
    attachInterrupt(0, wakeUpNow, LOW);
    sleep_mode();
 
    sleep_disable();
}
 
void loop() {
  Serial.print("Awake for ");
  Serial.print(count);
  Serial.println("sec");
  count++;
  delay(1000);

  if (count >= 5) {
      Serial.println("Timer: Entering Sleep mode");
      Serial.flush();
      delay(100);
      
      count = 0;
      sleepNow();
  }
}
```

leozulfiu, 2018-03-21:
I added another println statement (`Serial.println("In setup method");`) in the setup method to debug the issue and found out that the microcontroller is restarting itself after each sleep cycle.
Is this a bug or an issue with my sketch?

awatterott, 2018-04-14:
I have tested your program with IDE 1.8.5 + AVR Boards 1.6.206  and it is working. I have only activated the pull-up on pin 2.
```
void setup() {
  pinMode(wakePin, INPUT);
  digitalWrite(wakePin, HIGH);
```

closed 2018-04-20 by awatterott