watterott/ATmega328PB-Testing #35: High power consumption in deep sleep

Mirdox, 2018-05-06:
Hello,

Thank for this nice project.
I'm experimenting with 328pb using a very simple setup, just atmega328pb, resonator @8MHz, capacitors and reset pull-up.

It seems to have quite high power consumption in deep sleep.

atmega328pb has 19.1 uA, while using the same script atmega328p has 4.6 uA

I've checked the registers and they seem to be the same, except the PRUSART1 register.

```
  noInterrupts();
  ADCSRA &= ~(1 << ADEN);
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
  sleep_bod_disable();
  interrupts();
  sleep_mode();
  sleep_disable();
  ADCSRA |= (1 << ADEN); //Enable ADC
```

Am I doing something wrong?

Thanks for the ideas.

joysfera, 2018-06-01:
FYI, this is classical Low Power library but adapted for **328PB**:
https://github.com/canique/Low-Power-Canique/

As for the 328PB itself, I have measued 2.9 µA (with RFM69 attached, so without that it would be less than 2.8 µA).

Mirdox, 2018-06-08:
Great!
With this library I've got 3.3 uA, just with resonator.

Thank you very much.

Diky Petr

closed 2018-06-23 by Mirdox