watterott/ATmega328PB-Testing #38: Upload bootloader fail

maxpromer, 2018-06-19:
Hello !, I have ATmega328pb-au and i want to upload arduino bootloader. five chip i try to upload bootloader and there have same error.

```
C:\Users\Max\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude.exe -CC:\Users\Max\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.2/tools/avrdude.conf -v -patmega328pb -cstk500v1 -PCOM3 -b19200 -e -Ulock:w:0xFF:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude.exe: Version 6.3-20171130
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\Max\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.2/tools/avrdude.conf"

             Using Port                    : COM3
             Using Programmer              : stk500v1
             Overriding Baud Rate          : 19200
             AVR Part                      : ATmega328PB
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
             Hardware Version: 2
             Firmware Version: 1.18
             Topcard         : Unknown
             Vtarget         : 0.0 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x1e9516 (probably m328pb)
avrdude.exe: erasing chip
avrdude.exe: reading input file "0xFF"
avrdude.exe: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude.exe: 1 bytes of lock written
avrdude.exe: verifying lock memory against 0xFF:
avrdude.exe: load data lock data from input file 0xFF:
avrdude.exe: input file 0xFF contains 1 bytes
avrdude.exe: reading on-chip lock data:

[... 113 lines cut ...]
avrdude.exe: erasing chip
avrdude.exe: reading input file "C:\Users\Max\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.2/bootloaders/optiboot_m328pb.hex"
avrdude.exe: writing flash (32748 bytes):

Writing | ################################################## | 100% -0.00s

avrdude.exe: 32748 bytes of flash written
avrdude.exe: verifying flash memory against C:\Users\Max\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.2/bootloaders/optiboot_m328pb.hex:
avrdude.exe: load data flash data from input file C:\Users\Max\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.2/bootloaders/optiboot_m328pb.hex:
avrdude.exe: input file C:\Users\Max\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.2/bootloaders/optiboot_m328pb.hex contains 32748 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% -0.00s

avrdude.exe: verifying ...
avrdude.exe: verification error, first mismatch at byte 0x7e00
             0x00 != 0x1f
avrdude.exe: verification error; content mismatch

avrdude.exe done.  Thank you.

Error while burning bootloader.
```

Thank for replay.

awatterott, 2018-06-19:
What type of programmer are you using?
`SCK period: 0.1 us` is 10MHz SPI frequency, which would be too fast.

maxpromer, 2018-06-19:
I try USBasp and Arduino Uno (Arduino Uno R3 upload ArduinoISP sketch). How to change SPI frequency ?

awatterott, 2018-06-21:
If you are using the ArduinoISP sketch then SPI frequency is slow enough.
Maybe it is a hardware problem. What board with ATmega328pb are you using? If it is a selfmade board, then compare it with our [Wattuino Pro Mini PB](https://github.com/watterott/Wattuino/blob/master/hardware/Wattuino-Pro-Mini-PB_v10.pdf).

closed 2018-07-07 by awatterott