watterott/ATmega328PB-Testing #40: PlatformIO support

streamliner18, 2018-06-26:
It'll be really great if you guys can make a platform package for PlatformIO, as the 328PB support from their official toolchain looks broken right now. Much appreciated!

SergiuToporjinschi, 2019-08-25:
Ohh yes, I need this too. Would be very nice to have it in platformio... currently I have all projects on platformio and only one in arduino because of this....

SergiuToporjinschi, 2019-08-25:
@streamliner18 I found it on platformio:
this are the settings that worked for me 
[env:atm328p]
framework = arduino
platform = atmelavr
board = ATmega328PB

; change microcontroller
board_build.mcu = atmega328pb

; change MCU frequency
board_build.f_cpu = 16000000L

; upload_port = COM15
monitor_speed = 19200
upload_protocol = stk500v1
upload_flags =
    -F
    -P$UPLOAD_PORT
    -b$UPLOAD_SPEED

ritesh006, 2025-08-25:
is this issue still relevant

haimiko, 2025-08-25:
I've been using it like this and it works with this configuration.

[env:ATmega328PB]
platform = atmelavr
platform_packages = toolchain-atmelavr@3.70300.220127
lib_extra_dirs = ./lib
board = ATmega328PB
board_build.f_cpu =  8000000L
; change microcontroller
board_build.mcu = atmega328pb

ritesh006, 2025-08-25:
if issue solve you should open pr request. May be after that owner would like to review

awatterott, 2025-08-25:
PlatformIO has ATmega328PB support, but I dont know if the additional libraries are available.
I have now closed the issue.