watterott/ATmega328PB-Testing #43: Bootloader and Sketch Upload with ISP works, Serial doesn't

GoingForGold13, 2018-07-20:
Hi all,
I have a board with an Atmega328PB. Upload of the bootloader works, if I upload the sketch with ISP also works. However, if I flash the bootloader, I still can't get the Serial communication and programming to work. It is 3.3v and has a 8Mhz crystal on the board I'd like to use.

Bootloader:
`
avrdude.exe: Version 6.3-20171130
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\LocalUser\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.3/tools/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbtiny
avrdude.exe: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
             AVR Part                      : ATmega328PB
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

             Programmer Type : USBtiny
             Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude.exe: programmer operation not supported

avrdude.exe: Using SCK period of 10 usec
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9516 (probably m328pb)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: Using SCK period of 10 usec
avrdude.exe: reading input file "C:\Users\LocalUser\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.3/bootloaders/optiboot_m328pb.hex"
avrdude.exe: writing flash (32748 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 32748 bytes of flash written
avrdude.exe: verifying flash memory against C:\Users\LocalUser\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.3/bootloaders/optiboot_m328pb.hex:
avrdude.exe: load data flash data from input file C:\Users\LocalUser\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.3/bootloaders/optiboot_m328pb.hex:
avrdude.exe: input file C:\Users\LocalUser\AppData\Local\Arduino15\packages\m328pb\hardware\avr\1.1.3/bootloaders/optiboot_m328pb.hex contains 32748 bytes
avrdude.exe: reading on-chip flash data:

Reading | ################################################## | 100% -0.00s

[... 104 lines cut ...]
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "C:\Users\LOCALU~1\AppData\Local\Temp\arduino_build_733690/sonde_0007.ino.hex"
avrdude: writing flash (19430 bytes):

Writing | ################################################## | 100% 11.40s

avrdude: 19430 bytes of flash written
avrdude: verifying flash memory against C:\Users\LOCALU~1\AppData\Local\Temp\arduino_build_733690/sonde_0007.ino.hex:
avrdude: load data flash data from input file C:\Users\LOCALU~1\AppData\Local\Temp\arduino_build_733690/sonde_0007.ino.hex:
avrdude: input file C:\Users\LOCALU~1\AppData\Local\Temp\arduino_build_733690/sonde_0007.ino.hex contains 19430 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 10.18s

avrdude: verifying ...
avrdude: 19430 bytes of flash verified

avrdude done.  Thank you.`

Sorry, I put everything within code tags, but it somehow does not seem to work correctly...

Any idea what I could check?

awatterott, 2018-07-20:
Hi,
Programming looks good, so
* Is the 328PB and the serial adapter both 3.3V? Because the 328PB has a [different reset level](https://github.com/watterott/ATmega328PB-Testing#reset) than the 328.
* Is the serial wiring correct, Rx to Tx and Tx to Rx?

GoingForGold13, 2018-07-20:
Thanks!

I double checked the serial connection. Regarding the reset level - what is your suggestion to be able to handle the different level?

awatterott, 2018-07-21:
The reset voltage level is only a problem, if the 328PB is powered with 5V and the serial adapter uses 3.3V logic levels.
Do you use the standard Arduino reset circuit with [10k pull-up and 100nF in series](https://sites.google.com/site/wayneholder/_/rsrc/1518305388741/debugwire3/Arduino-RESET.png)?

closed 2018-08-05 by awatterott