watterott/ATmega328PB-Testing #44: Dead chips

LuisCRSousa, 2018-07-20:
After tried to upload this bootloader to my custom 3.3V and external 8MHz  the chip loats connection to the ISP MKII and ICE. I can't get it even to read the fuses using the atmel studio tools

awatterott, 2018-07-21:
How do you have programmed the bootloader? If you have used the Arduino IDE, which board have you selected?
Please also post a schematic of your custom board.

LuisCRSousa, 2018-07-24:
Programmed with Arduino IDE and selected 328pb 3.3V 8Mhz external. Tested with 3 chips, all dead after bootloader. All worked fine before the upload.

awatterott, 2018-07-25:
For a crystal oscillator or ceramic resonators you have to select "Crystal Clock".
"External Clock" is for an active ext. clock source.

closed 2018-08-05 by awatterott