watterott/ATmega328PB-Testing #45: Optiboot PB loader on Nano Clone no worky

zdravke, 2018-11-02:
I've burnt optiboot_m328PB.hex with avrdude on Nano clone with Atmega328PB with another Nano as ISP and there were no errors. However, nothing works. Tried optiboot_m328pb_blink.hex which also burns fine, but led does not blink. Burning back m328P bootloader works fine. 

Arduino IDE also cannot upload anything on PB variant even though Wattuino boards package was installed. Any ideas what I'm doing wrong?

zdravke, 2018-11-03:
Thanks, it works now:) I am now trying to test 16bit PWM, but that is not working now, even though the same sketch from here worked with m328P:

https://www.arduinoslovakia.eu/blog/2017/7/16-bitove-rozlisenie-pwm-pre-arduino?lang=en

zdravke, 2018-11-03:
Thanks, it works now:) Do you know how to send 16bit PWM to other pins, except for pin 9 and 10?

awatterott, 2018-11-03:
Have a look in the datasheet for OCxx pins. These are output compare pins from the timers and can generate PWM signals.

closed 2018-11-03 by awatterott