watterott/ATmega328PB-Testing #46: Atmega328PB and RFM95W

Lanzones, 2018-11-05:
Hi,
I have a lora GPS shield that works perfectly with the UNO with atmega328P. I have then developed a board with 328PB instead. I was able to load the bootloader and the sketch thru ISP. Recycled power and received an error on my terminal with the message below. I have tried the monitoring the SPI from micro to the RF module but it seems that there was no activity on all spi lines. I have recheck my lmic pin maps plenty of times and it has same pin assignment as my 328P development board with the lora gps shield. 
Just wondering if someone has already use this repo and compiled it with the lmic for lora modules. I might be missing something very simple. 
********************
LoRa GPS Example---- 
Connect to TTN
FAILURE 
C:\Users\User\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689
*********************
Please help..

awatterott, 2018-11-05:
Do you use this board package for Arduino IDE? If yes, try the examples from the SPI library to check if it is working and if it is a software or hardware issue.

Lanzones, 2018-11-08:
Sorry my mistake.. I probe once again the SPI pins to the RFM95W and I have attached the plots.. CS waveform should be acceptable I believe and the MISO only outputted a bit at the 2nd byte.. but still doesnt work though. The CLk by the way is around 5MHZ

I was thinking if the CPOL or LSBfirst/MSBfirst has been changed on the 328PB version of the chip that I dont know about. I have tried tried slowing down the comms to 1MHz but still doesnt work.

below 5V means micro side and 3.3V is the RFM95 lora module.
[image]
ch3mosi_ch4cs
[image]
ch3mosi_ch4miso
[image]
ch3miso_ch4misomicro
[image]
ch3mosi_ch4clk

awatterott, 2018-11-08:
So SPI is working. You can flash an ATmega328(P) program to the ATmega328PB to verify that the hardware is working, because the chips are compatible in this way.
Further infos in the appnote [Differences between ATmega328/P and ATmega328PB](http://ww1.microchip.com/downloads/en/AppNotes/Atmel-42559-Differences-between-ATmega328P-and-ATmega328PB_ApplicationNote_AT15007.pdf).
I have closed the issue, because it has nothing to do with the boards package.

closed 2018-11-08 by awatterott