watterott/ATmega328PB-Testing #48: Device signature error while uploading bootloader

bhargavmakodia, 2018-12-12:
Made a custom PCB with minimum connections to drive Atmega328PB (SMD).
Connected 10k pull-up to reset, applied Vcc, Ground & external 16MHz crystal with 22pF load capacitors.

Currently  using IDE 1.8.7 with Arduino version 1.6.200 and avr 3.6.0

Connected the board correctly with ISP using Arduino r3 as programmer. 1st uploaded ArduiniISp to the programmer and then wired ATmega328PB using ISP. Selected Atmega328PB internal crystal and 8MHz in drop down menu. Clicked bootloader
Shows below error:

`avrdude.exe: Device signature = 0x000000`
`avrdude.exe: Yikes!  Invalid device signature.`

awatterott, 2018-12-13:
There should be no capacitor in series with the reset signal for programming via ISP. This is only needed for serial programming via the bootloader.

bhargavmakodia, 2018-12-13:
It worked!
but I'm unable to upload program via ISP! So I used USB - UART CP2102 bridge module.

it uploads program only once. On programming 2nd time I need to connect ISP & flash bootloader to program via UART.

awatterott, 2018-12-13:
I would suggest to check the reset circuit, maybe the bootloader is not started via UART. For testing you can also try a reset button and press it before uploading.
I have closed the issue because it is not related to the ATmega328PB boards package.

closed 2018-12-13 by awatterott