watterott/ATmega328PB-Testing #50: Problems with simultaneous master/slave

haimiko, 2019-12-14:
Hangs when I try to run as an I2C slave on SDA0/SCL0 and Master on SDA1/SCL1.  Everything appears to initialize properly.  When I run Wire.begin(8) my remote master can see this device however, as soon as I query a slave device on SDA1, it no longer responds as a slave on SDA0.

awatterott, 2019-12-15:
The Wire1 lib deactivated the pullups from SDA0 and SCL0. I have fixed it now and I will release a new boards package soon.
https://github.com/watterott/ATmega328PB-Testing/commit/3ed2913c79806d61b47964d342f125fa49a116ca

haimiko, 2019-12-15:
> 
> 
> The Wire1 lib deactivated the pullups from SDA0 and SCL0. I have fixed it now and I will release a new boards package soon.
> Commit: [3ed2913](https://github.com/watterott/ATmega328PB-Testing/commit/3ed2913c79806d61b47964d342f125fa49a116ca)

Nice catch but still didn't resolve the issue.  Must be some other reference somewhere too.

awatterott, 2019-12-15:
I have looked over the code again, but I dont see any problems.
Can you make a minimal example to re-produce the problem.

closed 2020-01-22 by awatterott