watterott/ATmega328PB-Testing #52: SPI1

wangnick, 2021-01-06:
Hi Andreas,

why is there SPI1.h and SPI1.cpp implementing new classes SPI1Class and SPI1Settings, and then one new extern instance SPI1?

Why not rather redesigned SPI.h and SPI.cpp implementing SPIClass and SPISettings such that two extern instances SPI and SPI1 are supported as members of class SPIClass and thus acceptable to every other library out there, similar to what was done with Serial1, Serial2 etc ???

I'm trying to use the Atmega328PB board together with an MCP2515 CAN bus module AND two UARTs AND I need two ICPs, so I need to evade to SPI1 for the MCP2515 connection. I managed to adapt my CAN bus boot loader, but now I struggle with my application and with the ACAN2515 library of Pierre Molinaro and the stuff I added on top ...

Kind regards,
Sebastian
--
Sebastian Wangnick

awatterott, 2021-01-06:
Sure, that is possible and we are open for improvements. Its also the same on the Wire library.

wangnick, 2021-01-07:
Hi Andreas,

I've changed SPI.h and SPI.cpp as indicated above, see at https://github.com/wangnick/ATmega328PB-Testing. No fancy template programming, so I had to introduce some member variables and I had to make the methods non-static ...

Kind regards,
Sebastian